kylemanna/sniffer #1: TTGO and PCB spacing

mckochan, 2021-03-26:
To anyone that has assembled one of these:
From the pictures, it looks like the TTGO either didn't have the battery connector or you removed it to allow the boards to sit close enough to use just the spacers on the male header pins?

Is there spacers available that I can use to raise the TTGO or does the battery connector removed easily?

Thanks in advance for this awesome project!

kylemanna, 2021-03-26:
I mounted the headers so they are slightly raised above the PCB to clear the battery connector.

For example if you took something small (toothpicks?) and put it in under the headers when you solder the 0.1" headers it'll solve the problem. Remove the toothpick (or whatever) after soldering.  I think I actually use a spare PCB from one side to hold the header at a consistent height.

See picture:
[image]

mckochan, 2021-03-26:
AH toothpick, genius! Thanks

closed 2021-03-26 by mckochan