kylemanna/sniffer #2: High temp value

Ralphy140, 2021-07-11:
comparing the value of other sensors sitting next to the Sniffer they seem to over read by 2-3C I suspect that this might be caused by the internal VOC sensor being on all the time as opposed to taking a sample every 60 seconds though i may be wrong. 

In regards to the VOC and Air quality i have also found this that indicates how to extract the Bosch drivers to derive the VOC air quality 

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME680-on-ESPHome-using-Bosch-s-BSEC-library/td-p/13210

I am no programmer but perhaps this could be updated into the code?

kylemanna, 2021-07-13:
I've noticed this as well and fear it may be higher then 3°C and have observed the same with the BME280 (no VOC sensor). I think it's actually being heated by the ESP32 through the pins.

I think the proper solution is to better isolate the BME680/280 better with a thermal PCB cutout or an external temp sensor like the [DS18B20](https://esphome.io/components/sensor/dallas.html?highlight=ds18b20)

I think the best workaround for this is to modify the esphome yaml code to provide a filter to subtract a 3°C offset or a more advanced function as I'm sure it's not a constant temperature offset.

Here's some documentation if you want to give it a try:
https://esphome.io/components/sensor/index.html#sensor-filters

Another option may be to reduce the power consumption of the ESP32 so it generates less heat, but I haven't looked to that yet.

Let me know if you get a chance to experiment with it!

kylemanna, 2021-07-13:
Another experiment would be to remove the BME680 from the pin headers and to attach it using wires that provide a poor thermal conduction path to see if this validates the idea.

Ralphy140, 2021-07-13:
Hey, 

I have updated the YAML to make the required changes to take advantage of the VOC components and also reduce the temp by 3C, Happy to make a pull request or send it thought to you if you would like?

Ralphy140, 2021-07-13:
Also it seems odd that it exactly 3C across your ones and the 3 that i have made,

Ralphy140, 2021-07-16:
Pull request created to update temp off-set and BME680 driver

closed 2021-07-16 by Ralphy140