kylemanna/sniffer #10: TTGO chip update

larrygoodstuff, 2022-06-21:
It looks like the TTGO boards with a CP2104 chips are no longer available. They are only now available with a CH9102F chip. After successfully putting one together 1+ years ago, I decided to do another 4 and bought the components. After soldering them together and powering up I just get an "SDcard mount fail" error. 

Will the pinouts need to change with the updated chip?

Awesome little project by the way.

hexonMD, 2024-01-20:
I ended up doing the same thing :(

SleepinDevil, 2025-10-23:
I am guessing no one has figured out how to get this project working with the new chip. It's a shame the new chip isn't backwards compatible, that feels weird for an ESP32 device especially!