kylemanna/sniffer #11: How do I upload the files to the TTGO?

zirdum, 2023-01-30:
Are there instructions on how to upload the yaml to the TTGO device?

Ralphy140, 2023-01-30:
This project uses ESPHome - https://esphome.io/ check out the getting started on the home page, It works really well with Home Assistant and would recommend you read up on that to get started. 

Just to be clear you dont "upload" the YMAL to the device, you use a tool called ESP home that turns the YAML into a compiled file that is then uploaded to the esp32, you never upload the yaml directly to the device, if you want to change it you need to re-complie it in esphome then re-upload the file

zirdum, 2023-01-31:
Thank you. I figured out how to compile the yaml and uploaded it successfully to the ESP32.

closed 2023-01-31 by zirdum