kylemanna/sniffer #14: 3D Printed Enclosure

raptor464, 2023-11-24:
I created a 3D print enclosure for this project. I actually made two of these sniffers. I'm thinking about uploading the files to Thingiverse. Let me know if you are interested in the STL files.
[image]
[image]

kylemanna, 2023-11-28:
Nice! Love to see it and thanks for sharing.  If you do upload it, please LMK so I can add links!

raptor464, 2023-12-03:
I put it up on Thingiverse:

https://www.thingiverse.com/thing:6348428

duceduc, 2024-08-02:
Hi. Are you willing to sell them and ship it internationally? Also, I bought another kit to have it place outside my balcony. Any chance of creating another case for outside conditions? I don't need the lcd to be visible. Let me know Thanks .

chrisros, 2026-01-15:
I also made one: https://makerworld.com/en/models/2249235-2bluesc-sniffer-v1-0-0-lilygo-t-display-esp32#profileId-2449299