kylemanna/sniffer #15: Pin 4 Used in Multiple Places

bepstein111, 2023-12-21:
Unsure if this is a bug that ESPHome introduced, but when I went to update my sniffer with a new version, it complained that "Pin 4 is used in multiple places," both in as a gpio switch and as the backlight pin for the st7789v. Are you aware of this?

Ralphy140, 2023-12-22:
This is a new feature to stop misconfiguration where you reuse a pin by accident, It seems that this might be an edge case they have not captured in the feature release of 2023.12

Ralphy140, 2023-12-22:
Seems like someone has lodged a bug for this and the answer is to add Add "allow_other_uses: true" 

I have not updated to 2023.12 yet but I would suggest adding it to the switch and see what happens

Ralphy140, 2023-12-22:
switch:
  - platform: gpio
    pin: GPIO4
    name: "${upper_devicename} Backlight"
    id: backlight
    restore_mode: ALWAYS_ON
    allow_other_uses: true"

bepstein111, 2023-12-22:
`[allow_other_uses] is an invalid option for [switch.gpio]. Please check the indentation.`

hmmmm

Ralphy140, 2023-12-27:
Ok had a play, The code needs to be updated as following 

'''
switch:
  - platform: gpio
    pin: 
      number: GPIO4
      allow_other_uses: true
    name: "${upper_devicename} Backlight"
    id: backlight
    restore_mode: ALWAYS_ON'''

and 

'''
    backlight_pin:
      number: GPIO4
      allow_other_uses: true'''

Ralphy140, 2023-12-27:
This will need to be rolled into the main code by @kylemanna going forward

bepstein111, 2023-12-29:
Yep, just figured this out myself, they finally seem to have added it to the documentation.
For reference: https://esphome.io/guides/configuration-types.html?highlight=allow_other_uses#pin-schema

closed 2023-12-29 by bepstein111

bepstein111, 2024-01-02:
I'll throw a pull request together.