kylemanna/sniffer #18: New issues due to ESPhome updates

jruben4, 2024-03-01:
sniffer_common has errors now with the new version of ESPhome due to the new way ESPhome wants you to program shared pins (pins defined in more than one place).

Ralphy140, 2024-03-01:
Has been resolved in issue 15 and new code has been merged into this repo

jruben4, 2024-03-01:
Trying the new code.  Getting error "Couldn't find ID 'temp2'." in sniffer_common.

michaellunzer, 2024-04-27:
hey @jruben4 same "temp2" error here. any suggestions?

jruben4, 2024-04-27:
> hey @jruben4 same "temp2" error here. any suggestions?

Yes, I edited sniffer_common.yaml to take out all the temp2 references.  Here it is:

`esphome:
  name: $devicename
  platform: ESP32
  board: featheresp32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${upper_devicename} Hotspot"
    password: "lost_sniffuh"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_api_key

ota:
  password: !secret ota_password

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

# Grab font from https://fonts.google.com/specimen/Roboto and place in same directory as this file
font:
  - file: "Roboto-Medium.ttf"
    id: sz_24
    size: 24
  - file: "Roboto-Medium.ttf"
    id: sz_20
    size: 20
    glyphs: '!"%()+,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/³µΩ'
  - file: "Roboto-Medium.ttf"
    id: sz_13
    size: 13

binary_sensor:
  - platform: status
    name: "${upper_devicename} Node Status"
    id: system_status
[... 140 lines cut ...]
          } else if (id(pm_2_5).state < 350.4) {
            // hazardous
            id(pm_2_5_aqi).publish_state((400.0 - 301.0) / (350.4 - 250.5) * (id(pm_2_5).state - 250.5) + 301.0);
          } else if (id(pm_2_5).state < 500.4) {
            // hazardous 2
            id(pm_2_5_aqi).publish_state((500.0 - 401.0) / (500.4 - 350.5) * (id(pm_2_5).state - 350.5) + 401.0);
          }

    pm_10_0:
      id: pm_10
      name: "${upper_devicename} PM <10.0 µm"
      filters:
      - sliding_window_moving_average:
          window_size: 30
          send_every: 10

  - platform: template
    name: "${upper_devicename} PM <2.5 AQI"
    unit_of_measurement: "AQI"
    icon: "mdi:air-filter"
    accuracy_decimals: 0
    id: pm_2_5_aqi

  - platform: wifi_signal
    name: "${upper_devicename} WiFi Signal"`