kylemanna/sniffer #19: Failed config on sniffer.common.yaml during setup

hexonMD, 2024-03-02:
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/ttgo.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

display.st7789v: [source /config/esphome/sniffer_common.yaml:107]
  platform: st7789v
  model: TTGO_TDISPLAY_135X240
  backlight_pin: 
    number: 4
    allow_other_uses: True
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  cs_pin: 
    number: 5
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  dc_pin: 
    number: 16
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  reset_pin: 
    number: 23
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  update_interval: 1s
  rotation: 270
  
  Couldn't find ID 'temp2'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "temp".
  lambda: !lambda |-
    it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
    it.rectangle(0, 19, it.get_width(), it.get_height(), id(my_blue));   // header bar
    
    it.strftime(4,  3, id(sz_13), id(my_yellow), TextAlign::TOP_LEFT, "%Y-%m-%d %H:%M:%S", id(esptime).now());
    
    if (id(system_status).state) {
      it.print(236, 3, id(sz_13), id(my_green), TextAlign::TOP_RIGHT, "Sniffer0 Online");
    } else {
      it.print(236, 3, id(sz_13), id(my_red), TextAlign::TOP_RIGHT, "Sniffer0 Offline");
    }
    
    int r = 18;
    
    it.print(4,    r, id(sz_24), id(my_gray),   TextAlign::TOP_LEFT,  "PM2.5:");
    
    auto color = id(my_red);
    if (id(pm_2_5_aqi).state < 50) color = id(my_green);
    else if (id(pm_2_5_aqi).state < 100) color = id(my_yellow);
    it.printf(236, r, id(sz_24), color, TextAlign::TOP_RIGHT, "%0.0f AQI", id(pm_2_5_aqi).state);
    
    r += 21;
    it.print(4,    r, id(sz_20), id(my_gray),   TextAlign::TOP_LEFT,  "PM:");
    it.printf(236, r, id(sz_20), id(my_yellow), TextAlign::TOP_RIGHT, "%0.0f/%0.0f/%0.0f µg/m³", id(pm_1_0).state, id(pm_2_5).state, id(pm_10).state);
    
    r += 25;
    it.print(120,    r, id(sz_24), id(my_gray),   TextAlign::TOP_CENTER,  "Environment");
    
    r += 23;
    float temp_f = id(temp).state * (9.0/5.0) + 32.0;
    float temp2_f = id(temp2).state * (9.0/5.0) + 32.0;
    /*it.printf(4,   r, id(sz_20), id(my_yellow), TextAlign::TOP_LEFT, "%0.1f°F", id(temp).state); */
    it.printf(4,   r, id(sz_20), id(my_yellow), TextAlign::TOP_LEFT, "%0.1f°F", temp_f);
    it.printf(125, r, id(sz_24), id(my_yellow), TextAlign::TOP_CENTER, "%0.1f°F", temp2_f);
    it.printf(236, r, id(sz_20), id(my_yellow), TextAlign::TOP_RIGHT,  "%0.1f%%", id(humidity).state);
    
    r += 21;
    it.printf(4,   r, id(sz_20), id(my_yellow), TextAlign::TOP_LEFT,  "%0.1f hPa", id(pressure).state);
    it.printf(236, r, id(sz_20), id(my_yellow), TextAlign::TOP_RIGHT, "%0.1f kΩ",  id(gas_resistance).state/1000.0);
  auto_clear_enabled: True
  eightbitcolor: False
  height: 240
  width: 135
  offset_height: 52
  offset_width: 40

Ralphy140, 2024-03-02:
Check issue 15 this has been resolved

Ralphy140, 2024-03-02:
You will need to remove the refrence to temp2 if you are not using the ds18b temp sensor package 
```
      #float temp2_f = id(temp2).state * (9.0/5.0) + 32.0;
      /*it.printf(4,   r, id(sz_20), id(my_yellow), TextAlign::TOP_LEFT, "%0.1f°F", id(temp).state); */
      it.printf(4,   r, id(sz_20), id(my_yellow), TextAlign::TOP_LEFT, "%0.1f°F", temp_f);
      #it.printf(125, r, id(sz_24), id(my_yellow), TextAlign::TOP_CENTER, "%0.1f°F", temp2_f);
      it.printf(236, r, id(sz_20), id(my_yellow), TextAlign::TOP_RIGHT,  "%0.1f%%", id(humidity).state);

```